Pirate-MIDI/Pirate-MIDI-BridgeOS #12: Device locks up when trying to configure MIDI Thru

ChuckBarnard, 2022-05-14:
---

**Device(please complete the following information):**
 - Device: Bridge6
 - Firmware version: v1.0.1 both Beta 1 and Beta 2
 - Hardware version: [e.g. v1.0]

**Describe the bug**
When trying to access the MIDI thru settings the device locks up with all on the LEDs on red.  It changes the MIDI channel back to Omni.  This seems to have happened after changing the MIDI channel to something other then OMNI.  Even if I change it back to OMNI the issue still persists.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Menu - Global - MIDI - Thru Routing - TRS In or USB In

**Expected behavior**
Ideally it would be nice to be able to change the thru routing

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

samspencer5991, 2022-05-17:
Cannot replicate

codemonk84, 2022-05-27:
@ChuckBarnard Would it be possible to grab a video to show your experience from start to end?

ChuckBarnard, 2022-05-27:
Unfortunately I don't have a lot of process to show as I can't make it work.  It fails every time.  It doesn't look like I have the means to make a video small enough to attach.

ChuckBarnard, 2022-05-28:
It looks like the device not retaining it's MIDI channel may be a separate bug.  I just power cycle and it is back to Omni.

ChuckBarnard, 2022-06-02:
This appears to be working with Beta3

closed 2022-06-05 by samspencer5991