Pirate-MIDI/Pirate-MIDI-BridgeOS #21: TRS MIDI In

ChuckBarnard, 2022-06-02:
**Device(please complete the following information):**
 - Device: Bridge6
 - Firmware version: v1.0.1 Beta3
 - Hardware version:  v1.0

**Describe the bug**
TRS MIDI in has stopped working again on Beta3.

**To Reproduce**
Steps to reproduce the behavior:
Not receiving any messages from HX Stomp

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

samspencer5991, 2022-06-05:
Please provide some more context for this

ChuckBarnard, 2022-06-06:
I've updated the description based on our testing.

ChuckBarnard, 2022-11-16:
Not sure, I started playing at church again and had to move on with something else.  Just spent a little time poking around in the new web editor but ran into a couple of road blocks.  I'm not seeing a way to run a one time command at boot up.  Also, need to be able to toggle commands with the aux pedal connected to the FlexPort.

samspencer5991, 2022-11-16:
Ok cool, we'll we haven't had any other reports of MIDI in issues so I'll wait for your feedback. If you can give me some more info on these 'roadblocks' I can help sort them out.

Regarding messages on boot-up. It's not currently available. But a great idea. I'll bring that into the upcoming v1.2 update which is currently in closed beta testing.

When you say toggle commands, what exactly do you mean?

ChuckBarnard, 2022-11-16:
The two issues we are discussing now are the road blocks.

For the Aux pedal, I need one pedal to toggle on 127 and toggle off 0 the same effect.  It looks like I only have one press option.

samspencer5991, 2022-11-16:
Gotcha.

Well that's a fair bit beyond the scope of this issue on discord. But to address them quickly. I already mentioned the boot messages, and as for aux switch toggle messages, that's also coming in v1.2. We should have a public beta out next week :)