Pirate-MIDI/Pirate-MIDI-BridgeOS #36: Freezing after Serial communication

GriffinSauce, 2022-07-12:
**Device(please complete the following information):**
 - Device: Bridge6
 - Firmware version: 1.1.0-Beta7
 - Hardware version: 1.0.1

**Describe the bug**

After serial communication the device freezes, I've observed two behaviours:

**Issue 1: bank up/down, serial API freeze**
1. Factory reset
2. Send `0,CHCK~`
3. Device responds with device info
4. Bank down or up by pressing the relevant footswitches
5. Device response with `E,bankUp,7~`
6. Serial API stops responding until restart 

**Issue 2: footswitch, full freeze**
1. Factory reset
2. Send `0,CHCK~`
3. Device responds with device info
5. Press a single footswitch
6. Device responds with event
7. Footswitch lights up (sometimes) but the whole device stops responding until restart (no response from Serial API, footswitches, 1+4 long press or bank up/down switch combo's)

**Expected behavior**

For the device and API to stay responsive :D

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

I've encountered this with the JavaScript library but also confirmed the issue with the Arduino serial monitor and others. The sending method does not seem to matter.

Hardware:
- Macbook Air M1
- Macbook Pro intel
- Tried multiple cables, always directly connected

Like noted: I've done multiple factory resets, global settings resets, it doesn't matter what bank I am on.

GriffinSauce, 2022-07-12:
Update, some findings with previous beta's:
-  Beta 6:  same issues
- Beta 5:  worked initially but stopped after some testing and started behaving the same way. Reflashing firmware or factory reset did not fix it.

fourth44, 2022-07-24:
What I found on linux + JVM: if I use MidiSystem.getMidiDeviceInfo + MidiSystem.getMidiDevice and open an receiver and transmitter (i.e. 'connect to the midi device'), serial communication won't hang. If I leave the midi connection out, serial communication will definitely hang, especially after CTRL.

simonaglover95, 2022-08-27:
Is this still live, or needs closing?

GriffinSauce, 2022-08-30:
I'll check this again with the latest firmware tonight/tomorrrow.

GriffinSauce, 2022-08-31:
Yep, fixed in the latest beta! Thanks guys! 🙌

closed 2022-08-31 by GriffinSauce