Pirate-MIDI/Pirate-MIDI-BridgeOS #39: Bridge Crash | Flexport Midi Sync Pulse and LFO Message Stack Simultaneously

DanBradshaw, 2022-07-24:
**Device(please complete the following information):**
 - Device: Bridge6
 - Firmware version: 1.1.0_beta7
 - Hardware version: 101

**Describe the bug**
The Bridge Crashes when Flexport is set to Midi Synce Pulse and the LFO in on and assigned to a message stack. Brisge then requries a physical power cycle to resolve. Removing the flexport mode or the message stack on the LFO stops the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Assign Midi Synce Pulse to either Flexport
2. Enable LFO on a FS and assign a message stack
3. Exit edit mode and brisge will crash after a few seconds the LFO being enabled
4. *Probably easier to load up the attached json config

**Expected behavior**
Bridge doesn't crash

[bridge6_20220724.txt](https://github.com/Pirate-MIDI/Pirate-MIDI-Features-Bug-Tracking/files/9175307/bridge6_20220724.txt)

samspencer5991, 2022-08-01:
@DanBradshaw please provide more information on this. Is the LFO sync'd to a MIDI clock, if so which one? Or is it free running. What MIDI clock is the Sync Pulse using? Currently MIDI clocks require a switch configured as tap tempo to function, which switch is assigned etc....?

DanBradshaw, 2022-08-01:
Hey mate, LFO and sync pulse are assigned to clock A. FS3 is tap tempo. I attached the config.json from my bridge for you, probs easier to load it up for bench testing.

samspencer5991, 2022-08-01:
Yeah all good, I realised that just after I had a potato moment. I believe it's caused by a buffer overflow exception that wasn't handled properly.

closed 2022-08-02 by samspencer5991