Pirate-MIDI/Pirate-MIDI-BridgeOS #71: EDIT APP : Not Reading LFO Sync Mode and Frequency Settings from Device or File

DanBradshaw, 2022-08-20:
**Device(please complete the following information):**
 - Device: Bridge6
 - Firmware version: 1.1.0 Beta 8
 - Hardware version: 1.0.1
 - Windows 10
 - Chrome and Edge Browsers

**Describe the bug**
Whether importing from file or device, if an LFO is enabled on a switch the app doesn't appear to read Sync Mode and Frequency settings from json. 

This is loaded in:
```
                    "name": "Filter",
                    "primaryState": false,
                    "secondaryState": false,
                    "primaryMode": "toggle",
                    "secondaryMode": "holdToggle",
                    "primaryColor": "29bc7d",
                    "secondaryColor": "00ffff",
                    "primaryLedMode": "onOff",
                    "secondaryLedMode": "onOff",
                    "sequentialPattern": "forward",
                    "sequentialRepeat": "all",
                    "sequentialSendMode": "always",
                    "linkedSwitch": 0,
                    "currentStep": 0,
                    "midiClock": null,
                    "lfo": {
                        "state": false,
                        "frequency": "1/4",
                        "minLimit": 0,
                        "maxLimit": 127,
                        "trigger": "primary",
                        "messages": "press",
                        "waveform": "sine",
                        "resolution": 8,
                        "resetOnStop": true,
                        "clock": 0
```
This is what the app shows when the above has been imported:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Switches, Switch x, LFO, State Enabled, Sync Clock A, Freq 1/4, Trigger Primary, Min 0, Max 127, Shape Sine etc
2. Click on Save
3. Confirm LFO is Toggling and working as expected
4. Open App and import form device
5. Inspect settings on Switch LFO is assigned too.
6. I've attached my config if you just want to load and check FS2 on Banks 0 and 1

[Bridge 6 - 4d002e4d00324d0036 (6).txt](https://github.com/Pirate-MIDI/Pirate-MIDI-Features-Bug-Tracking/files/9388059/Bridge.6.-.4d002e4d00324d0036.6.txt)

Konnsalad, 2022-08-25:
Just pushed a fix to this, can you give it a test and confirm that it's working as intended now.

DanBradshaw, 2022-08-25:
Looks good mate! Cheers

closed 2022-08-25 by Konnsalad