Pirate-MIDI/Pirate-MIDI-BridgeOS #74: EDIT APP: Retain link between device library and command

GriffinSauce, 2022-08-21:
**Note: this is a feature request.**

**Issue**
After selecting a command through the device library, the raw midi command is added to the config and the context from the device library is lost.

**Example**
- open Switch messages
- click "+ Device library"
- select Line 6 HX Stomp -> Footswitch 1
- the command `Control Change - [Ch: 1] [Controller: 49] [Value: 127]` is added
- editing the command brings up the Control Change parameters

**Desired/expected outcome**
- the command shows the parameters from the device library
- editing the command brings up the device library form

This way you can see what the command is actually doing and modify it in context. (in this example, change it to HX Stomp FS2 for example).

Right now the feature inserts a midi command and your options to modify it are:
1. learn the midi parameters of the device to modify it
  a. this defeats the purpose of device library
1. remove the command and add a new one
  a. this takes a lot of effort (and increases the potential for user error)
  a. introduces the need to remember parameters like the device channel

... assuming that you even remember which command is which later on. ;)

I think maintaining the abstraction of the device library would make a massive difference in the user experience.

Konnsalad, 2022-08-22:
Planned for implementation after the "Profiles" feature milestone as the API doesn't have room for additional command meta-data.

GriffinSauce, 2022-08-25:
Cool, is the plan to allow arbitrary meta-data ? 

That would be exciting, it would allow for all kinds of reversible abstractions!

Konnsalad, 2022-08-26:
Yeah that's the goal, still on the drawing board at this stage though.