Pirate-MIDI/Pirate-MIDI-BridgeOS #80: Tap tempo display not updating

simonaglover95, 2022-08-26:
**Device(please complete the following information):**
 - Device: Bridge6
 - Firmware version: V1.1.1.1-beta4
 - Hardware version: v1.1

**Describe the bug**
Tap Tempo scribble strip not updating with taps

**To Reproduce**
Steps to reproduce the behavior:
Factory reset
Change fs 1 to tap tempo
Exit menu and tap new tempo 
Led speed updates, but not the scribble strip. 

**Expected behavior**
When tapping the switch, the tempo should update live on the OLED screen.

simonaglover95, 2022-08-27:
When changing banks, display doesn't update, but entering menu and exiting again makes the tempo display change.

simonaglover95, 2022-08-30:
More details: readout doesn't update on Fs1. On Fs2, the tempo updates but seems to be about double the actual tapped tempo. Fs3, seems to be normal function.

samspencer5991, 2022-10-25:
@simonaglover95 is this fixed?

simonaglover95, 2022-10-26:
No, still there on FS1 after factory reset

samspencer5991, 2022-10-27:
Fixed in v1.1.2

closed 2022-10-27 by samspencer5991