Pirate-MIDI/Pirate-MIDI-BridgeOS #81: Expression Bank Messages on 2A not sending MIDI.

simonaglover95, 2022-08-27:
**Device(please complete the following information):**
 - Device: Bridge6
 - Firmware version: v1.1.1-beta4
 - Hardware version: v1.0.1

**Describe the bug**
When adding a bank message to expression pedal 2A, the message will not be sent.

**To Reproduce**
Steps to reproduce the behavior:
1. Change Flexiport 2 mode to Exp In
2. Add bank message to expression pedal 2A
3. MIDI monitoring app detects no MIDI when expression pedal is moved. 
4. (works as intended on Exp 1A)

**Expected behavior**
MIDI messages should be sent as intended when expression pedal is moved by the user.

frenno, 2022-10-07:
Hi @simonaglover95 and @samspencer5991,

I've been affected by this, I think it's probably the same bug as this but I can open a separate issue if you think it's a different one. Further info below.

I'm on the same firmware/hardware as the initial report (v1.1.1-beta4/v1.0.1).

My setup:

- Flexiport 1: Exp In

- Flexiport 2: Exp In

I added one Bank Message (CC) to Exp 1A.

I then added five Bank Messages (all CC) to Exp 2A.

At this point, Exp 1A works as expected. However Exp 2A only sends the message in position one (the pedal I was controlling didn't do what I expected, I then used a midi monitor on PC to verify that only this single message was being sent when I move Exp 2A).

I then added another Bank Message to Exp 1A, again this worked as expected. Surprisingly, the second message on Exp 2A was now being sent as well (I hadn't made any changes to the Exp 2A config/Bank Messages). After I added a third message to Exp 1A, the third Bank Message on Exp 2A was sent as well. And so on when four or five messages were added to Exp 1A.

So it seems Exp 2A is only sending as many messages as there are configured for Exp 1A.

Hopefully that narrows down the source of the bug.

If anyone else is affected by this, I've worked around it by adding "dummy" messages to Exp 1A:

- Go to Menu > Exp Pedals > Exp 1A > Bank Messages
- Press footswitch labelled "add"
- Use the footswitch labelled "->" to navigate to the second page of settings (you can leave the defaults on the first page)
- Set all outputs (FLEXI1, FLEXI2, DIN5, USB) to "OFF"
- Press footswitch labelled "save"
- Repeat the above as many times as necessary until you have at least as many messages in Exp 1A as in Exp 2A

Not ideal, but works fine for the moment.

samspencer5991, 2022-10-27:
@frenno this is extremely helpful. Thank you so much, fixing this now.

samspencer5991, 2022-10-27:
Fixed in v1.1.2

closed 2022-10-27 by samspencer5991