Pirate-MIDI/Pirate-MIDI-BridgeOS #88: Incorrect midi output

GriffinSauce, 2022-09-22:
**Device(please complete the following information):**
- Device: Bridge6
- Firmware version: v1.1.1.1-beta.4
- Hardware version: v1.0.1

**Describe the bug**
Incorrect midi messages are sent to the DIN5 port (not to USB!) when switchOn / switchOff messages are used in particular combinations. It looks like a CC message is split into multiple PC messages somehow.

**To Reproduce**
Steps to reproduce the behavior:
1. Load the configuration attached below into a Bridge6
2. Press the various switched, the bug appears on switch 2 and 3, especially when switching between them
3. Monitor the midi output with an external device (again, **not** via USB)

**Expected behavior**

The following set of messages is sent (with some varying values):
```
Channel 1 PC 2
Channel 2 CC 69 0
Channel 6 CC 81 0
(And switch 5 primary or secondary is enabled)
```

**Actual behavior**
```
Channel 1 PC 0
Channel 1 PC 127 <-- incorrect
Channel 1 PC 69 <-- incorrect
Channel 1 PC 0 <-- incorrect
Channel 6 CC 81 0
(And switch 5 primary or secondary is enabled)
```

**Additional context**
I've tried some things like reordering / adding /removing commands or footswitch configs, the only thing that consistently prevents the issue is removing the `switchOn` and `switchOff` messages. Anything else just moves the bug to different combinations of switches.

**Config JSON**
[debug-2022-09-22T20:03:30.110Z.json.renamethis.txt](https://github.com/Pirate-MIDI/Pirate-MIDI-Features-Bug-Tracking/files/9629138/debug-2022-09-22T20.03.30.110Z.json.renamethis.txt)
_(Github doesn't allow JSON attachments for some reason)_

GriffinSauce, 2022-09-24:
Some additional information: I noticed that if I add a "decoy" PC message in front with a different channel things become a bit clearer. The first message will be corrupted and "split" into several:

When the affected message is PC 3 the actual output is PC 0, PC 127, PC 3 to the same channel.

simonaglover95, 2022-10-27:
@GriffinSauce Is this still happening in the release firmware?

simonaglover95, 2023-01-27:
@GriffinSauce is this still happening in the latest fw v1.2.1?

simonaglover95, 2023-01-27:
@samspencer5991 does this need adding to our list for the next update?

simonaglover95, 2023-02-20:
@GriffinSauce is this still an issue?

closed 2023-03-22 by samspencer5991