Pirate-MIDI/Pirate-MIDI-BridgeOS #93: Sequential Steps are buggy

TheGuacamoleXplosion, 2022-10-23:
**Device:**
 - Device: Bridge4
 - Firmware version: 1.1.1
 - Hardware version: I guess 1.0? Firmware Update App says: UID 3d005a3d005e3d0062
 - https://edit.piratemidi.com/ running on current Chrome / Win10 system

**Describe the bug**
Not sure whether this is just one, or several, bugs: 
Sequential Steps are out of sync. LED colours behave as expected, other step messages are one step behind.
Also, a Step 1 populated with no messages at all will apparantly send CC#00 with value 127 on all outputs.
Display of steps is not perfect; no descenders (“Play” becomes “Plau”), and the upcoming steps displayed in the detailed display view are always in linear order, rather than showing the actually upcoming next one (in case the sequential loop repeats only the last steps and not all).

**To Reproduce**
Steps to reproduce the behavior:
1. Create a blank Bridge4 template in the app. All MIDI PC send off in global settings. Also, call Bank 0 “TL Looper” just for kicks.
2. Delete pre-populated Toggle On/Off settings for the Switches (blank template is not REALLY blank BTW)
3. Set Switch 1 to sequential, forward, last 2 steps repeating. 
4. Create a 4 step sequence as pictured in screenshot “SequentialMess.jpg”, all CCs sent only to TRS. (BTW, the concrete CC sequence here is just for testing; actual sequence for controlling the Strymon Timeline sequencer would be different)
5. Send to Device. Hit Switch 1 repeatedly. This will yield the results displayed in the MIDI-OX input monitor in “SequentialMess.jpg” 

So, what apparently happens:
- Step 1 is the initial step: The basic behaviour when entering a bank with this sequential step (you might consider renaming it to Step 0, thus). This step has the correct LED colour as defined in sequential Step 1. MIDI-message-wise, however, it does nothing. Which is kinda okay, since Step 1 is the initial state (and if you want bank initial behaviour, you can program bank messages for that).
- Step 2 is what happens after hitting the Switch for the first time. LED behaviour is correct again. MIDI message send is however the message programmed for Step 1. Or in case where no message at all is programmed for Step 1, CC#00 with value 127 will be sent on all outputs (at least via TRS and USB; I did not monitor the Flexiports). 
- Step 3 has again correct LED colour, and sends Step 2 messages.
- Step 4 correct LED, sends Step 3 message.
- Step 4 message will never be sent.

**Expected behavior**
Each step sends its assigned MIDI messages. Maybe there should be a clearly labelled Step 0 that can only have name and colour, but no messages.
Step labels are displayed with descenders.
Upcoming step label in detailed view shows actually upcoming step (not just the one which would follow in a linear loop across all steps). 

**Screenshots**
[image]

TheGuacamoleXplosion, 2022-10-23:
Addition: This seems to be an App issue rather than a device issue. I exported this sequence (which would result in the desired behaviour for controlling the Strymon Timeline looper):

[image]

When I edit that sequence on the device itself, I can see that there, the MIDI messages are displayed within the step where they are actually sent. Reimporting this to the app also shows this:

[image]

ninety7, 2022-10-29:
Same issue here, I can't even change any other switches because every time I send data to the bridge6, all sequential configurations get screwed up

samspencer5991, 2022-10-31:
I discovered that when receiving a sequential config from the editor, sequential messages were offset incorrectly. This appears fixed in v1.1.3 for me, but it would be good to get your confirmation.

closed 2022-10-31 by samspencer5991