Pirate-MIDI/Pirate-MIDI-BridgeOS #94: Certain Messages cannot be saved in the App (MIDI notes and Smart Messages in certain contexts)

TheGuacamoleXplosion, 2022-10-23:
**Device(please complete the following information):**
 - Device: Bridge4
 - Firmware version: 1.1.1
 - Hardware version: I guess 1.0? Firmware Update App says: UID 3d005a3d005e3d0062
 - https://edit.piratemidi.com/ running on current Chrome / Win10 system

**Describe the bug**
Certain Messages cannot be saved to certain MIDI message stacks in the apps. Like
- MIDI note commands from the Device Library apparently cannot be saved at all in the App.
- Smart Messages can be saved in the App for Toggle Messages, but not for Sequential Steps. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create a blank Bridge4 template in the app.
2. Delete pre-populated Toggle On/Off settings for the Switches.
3. Click Switch 1, Toggle On, Device Library. Select Effects>Strymon>Timeline>Play (MIDI Notes) – or any other MIDI Note command. Click on save. Will not work.
4. Now, change switch mode to Sequential (always, foward, all). Create a Step. Add a Smart Message, e.g. Switch On/Switch 2/primary. Click save. Nothing is saved, and now, you cannot even save the empty step anymore. You can only cancel.

**Expected behavior**
MIDI-note based commands from the Device Library can be saved in the App.
Smart Messages can be saved to Sequential Steps in the App.

TheGuacamoleXplosion, 2022-10-23:
Addition: 
I did succeed in adding a Smart Message to a Sequential Step on the device itself. It works as expected. When I import this into the editor, the Smart Message is displayed correctly (see screenshot ImportedSequence.jpg); but I cannot edit any of the steps in the app now, clicking on the pen symbol will not work.
[image]

samspencer5991, 2022-10-27:
@Konnsalad

closed 2023-01-27 by simonaglover95