Pirate-MIDI/Pirate-MIDI-BridgeOS #95: Preserve Switch Toggle States should also affect Sequential states

TheGuacamoleXplosion, 2022-10-23:
**Device(please complete the following information):**
- Device: Bridge4
- Firmware version: 1.1.1
- Hardware version: I guess 1.0? Firmware Update App says: UID 3d005a3d005e3d0062
- https://edit.piratemidi.com/ running on current Chrome / Win10 system

**Describe the bug**
In the very strict sense, this is not a bug, but an enhancement request: The global parameter is called “Preserve Switch Toggle States”, and it does exactly this.
However, in a scenario where you use one bank of the Bridge4/Bridge6 for control of a permanently active looper (such as the loopers in a Strymon Timeline or Line6 M9 / M13), and you have set up sequential switches like “Record – Overdub – Play”, you will want that switch to remember the current looper state even if you change banks to do other MIDI stuff. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create a sequential Switch with e.g. 3 Steps.
2. Set global parameter “Preserve Switch Toggle States” = yes.
3. Go to the 3rd step of the sequential switch. 
4. Switch bank and switch back. You will see that the sequential switch will have reset to step 1.

**Expected behavior**
Either the global parameter “Preserve Switch Toggle States” should include preserving sequential steps; or there should be an additional global parameter “Preserve Switch Sequential Steps”.

grwxyz, 2023-02-04:
This will be a useful enhancement!

grwxyz, 2023-07-04:
Any word on whether this will be included in an upcoming update? Thanks again for all the incremental improvements!

simonaglover95, 2024-05-28:
Fixed in BridgeOS 2.0 Release

closed 2024-05-28 by simonaglover95