Pirate-MIDI/Pirate-MIDI-BridgeOS #100: When using Go To Bank smart message, message stacks/LEDs in destination bank are triggered/activated

frenno, 2022-10-29:
**Device(please complete the following information):**
 - Bridge6
 - Firmware version: 1.1.2
 - Hardware version: 1.0.1

**Describe the bug**
Switch states are not reset when entering a new bank, so the same footswitch in the new bank (and associated message stacks/LEDs) are activated.

**To Reproduce**
Steps to reproduce the behavior:
1. Start with a blank template and Bank Change PC messages turned off to eliminate any messages not set up below.

2. On Bank0/FS1, add the following messages to the specified stacks:
    - Toggle On message (eg. CC 16 127 127)
    - Press message (eg. CC 15 126 126)
    - Release message (eg. CC 14 125 125)
    - Hold message (eg. CC 13 124 124)
    - Hold Release message (eg. CC 12 123 123)
    - Secondary Toggle On (eg. CC 11 122 122)

3. (a). On Bank1/FS1, add a Toggle On message Go To Bank 0
4. (a). Navigate to Bank 1 on the device and tap FS1

<ol start=3>
<li>(b). On Bank1/FS1, delete the Toggle On message and add a Press message Go To Bank 0</li>
<li>(b). Navigate to Bank 1 on the device and tap FS1</li>
</ol>

<ol start=3>
<li>(c). On Bank1/FS1, delete the Press message and add a Hold message Go To Bank 0</li>
<li>(c). Navigate to Bank 1 on the device and hold FS1</li>
<li>Observe that in the various steps 3 and 4, various message stacks are triggered in Bank 0 even though the footswitch was pressed down in Bank 1. Also when the Go To Bank message is in a Press stack, the LED of the switch in the new bank is turned on.</li>
</ol>

**Expected behavior**
When a footswitch with a Go To Bank message is pressed, the device should change to that bank and not do anything else. Any input from that footswitch should be ignored until after it is released (and then honoured for subsequent presses).

**Screenshots**
In the screenshot (of the Protokol MIDI monitor), I added a marker in between a/b/c versions. You can see that the different message stacks are triggered depending on Bank Jump times relative to switch press/release/hold.

**Additional context**
- the messages above aren't important, what is important is that the various message stacks are being triggered (along with the LEDs)
- this was with the "Preserve Switch Toggle States" setting off, so I didn't test whether the Toggle Off stacks can be activated.
- same results with Bank Down smart message (didn't try Bank Up).
- I left all switch configs as defaults but it also happens with eg Hold Momentary as secondary config for Bank0/FS1
[image]

frenno, 2022-10-29:
[Bridge6SwitchBug.txt](https://github.com/Pirate-MIDI/Pirate-MIDI-Features-Bug-Tracking/files/9893837/Bridge6SwitchBug.txt)

This is a Bridge6 config file with all steps to 2a (Github doesn't let me attach JSON so you'll have to rename the extension)

samspencer5991, 2022-10-31:
addressed with v1.1.3

closed 2022-10-31 by samspencer5991

frenno, 2022-11-01:
[image]

Unfortunately this doesn't seem to be fully addressed in 1.1.3, I tested using the same steps as above plus one other variant.

- If I use the Toggle On stack on Bank1/FS1 for the Go To Bank message, the Toggle On stack in the destination bank is still sent (first message in screenshot)
- If using the Press stack, this appears to be fixed
- I can't get anything to send anymore using the Hold stack, whether it's a Go To Bank or regular MIDI message
- If using a Go To Bank message in the Secondary Toggle On stack, the Hold Release stack is sent
- If I change Bank0/FS1 secondary config to "Hold Momentary", and am still using a Secondary Toggle On message coming from Bank1, both the Hold and Hold Release stacks are sent (the two last lines on the screenshot)

samspencer5991, 2022-11-02:
Hmm ok, so it seems that the issue is separate with Go To Bank messages. I had only tested bank up/down, so I'll take another swing at this tomorrow.

frenno, 2022-12-12:
Hi @samspencer5991, currently using firmware 1.2.0-alpha.5 and still seeing all the same results as in my previous comment (which was using 1.1.3).