Pirate-MIDI/Pirate-MIDI-BridgeOS #106: Tap Tempo resetting when banks scrolled

Sopporiffic, 2022-11-03:
**Device(please complete the following information):**
 - Device: [Bridge 6]
 - Firmware version: [v1.1.3
 - Hardware version: [e.g. v1.1.2]

**Describe the bug**
Tap tempo isn't maintain when scrolling through banks

**To Reproduce**
Steps to reproduce the behavior:
1. Assign Tap tempo to a switch
2. Change Tempo using switch
3. Scroll bank in any direction
4. Scoll back to beginning bank
5. Tap tempo has reset to 120bpm.

**Expected behavior**
I was expecting midi clock A to maintain across banks. I also assigned tap tempo to switches across multiple banks and, at least visually, kept resetting to 120bpm as you scoll across banks

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

MarkFSI, 2023-02-23:
Hello,
I would only add that this problem also happen in my device, a Bridge6 HW v.1.1.3 and FW v.1.2.1 .
It would be a very good enhancement because this problem, in my opinion, makes the clocks useless if the user often changes bank

grwxyz, 2023-02-23:
I concur - this is a good enhancement. I have my Bridge setup with generic "effects specific" banks ie Gain, Time-Based, Modulation etc. Changing banks mid song is useful with the generic banks. The generic banks work really well for jamming and creating new songs. Currently I work around this limitation by creating song specific banks... this takes more time and also doesn't necessarily work if I'm in creative mode. Thanks!

MarkFSI, 2023-02-24:
Some new smart messages to set a specific bpm value and to send "start" and "stop" commands would be very useful.
I will request them shortly in a dedicated discussion

simonaglover95, 2024-05-28:
Fixed in BridgeOS 2.0 Release

closed 2024-05-28 by simonaglover95