Pirate-MIDI/Pirate-MIDI-BridgeOS #113: Editor stuck on "Synchronising Banks" at 24%

emilyperry456, 2022-12-18:
**Device(please complete the following information):**
 - Device: Bridge6
 - Firmware version: 1.1.1
 - Hardware version: v1.0.1

**Describe the bug**
In the editor app using Chrome version 108.0.5359.94 (Official Build) (x86_64) on 2018 Mac Mini, my device freezes and the editor cannot make it past 24% when "synchronising banks."

**To Reproduce**
Steps to reproduce the behavior:
1. Open the editor site
2. Click import from device and select Bridge6

**Expected behavior**
The initialization process will begin and the editor will get stuck at 24%.

samspencer5991, 2022-12-18:
Thanks for that @emilyperry456. We'll need a lot more information to help diagnose the problem though.

For example, what settings have been changed?

You're on an out dated firmware version. Have you tried updating the firmware first? Note that moving from v1.1 to v1.2 requires a factory reset in between.

samspencer5991, 2023-01-12:
@emilyperry456 any further updates on this or can I mark this as solved?

closed 2023-01-27 by simonaglover95