Pirate-MIDI/Pirate-MIDI-BridgeOS #119: Sequential step labels - Descender being cut off

grwxyz, 2022-12-29:
- Device: Bridge6
 - Firmware version: 1.2.0
 - Hardware version: 1.1.3

**Describe the bug**
When using characters with a descender (eg gipqy), the descender is being cut off (ie it doesn't display)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'edit.piratemidi.com', sign in and connect to device.
2. Create a 'Sequential' mode switch.
3. Add a 'New Step'
5. Add a label 'gipqy'
6. Save and send to device
7. Check the device display and notice that the bottom of the characters will appear cut off.

**Expected behavior**
The characters are displayed correctly.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
Add any other context about the problem here.

simonaglover95, 2023-01-27:
It's normal for the descender not to show on Extended UI, but in Simple UI we should be able to get this fixed in the next firmware update :)

grwxyz, 2023-01-27:
The only time the “descender” doesn’t show is when using a sequential switch on the bottom row (ie FS1/2/3). You can see the sequential label drops down a few pixels, therefore cutting off the “descender”.

Works on top row for all buttons.

Works on bottom row (ie FS1/2/3) for normal toggle buttons.

simonaglover95, 2023-01-27:
Yes, as I said it's only unintentional in Simple UI, which we shall get fixed very soon :)

closed 2023-02-20 by simonaglover95