Pirate-MIDI/Pirate-MIDI-BridgeOS #121: Unable to connect to the web editor

JiriHavlicek, 2023-01-06:
- Device: Bridge6
 - Firmware version: v1.2.0-beta.1
 - Hardware version: v1.1.3

The editor does not want to connect to Bridge6. When connecting for the first time, the error message "Error: Connecting and loading data failed - Serial Port Connecting Failed. Failed to execute 'open' on 'SerialPort': Failed to open serial port" appeared. The second attempt froze at 5% at 'Synchronizing global settings'.

I tried Factory reset, manual setting of SerialPort.

samspencer5991, 2023-01-06:
You're currently on the beta. Please update to the latest public build and try again

JiriHavlicek, 2023-01-06:
I am trying to update my device. It doesn't work according to the manual. I downloaded the firmware update app but the device is not recognized when connected. I am also trying it from multiple PCs. Nothing helps.

samspencer5991, 2023-01-06:
Ok, well that. Is the issue, not the editor. So let's solve that problem rather than this flow on effect.

JiriHavlicek, 2023-01-06:
OK, now the device has appeared in the Firmware updater application. An update version 1.2.1 was offered. I've had it update but everything seems to be stuck on "Waiting for device..."

JiriHavlicek, 2023-01-06:
uh, sorry - update version 1.2.0. but it still doesn't work

samspencer5991, 2023-01-07:
Can you please provide a screenshot of the error

JiriHavlicek, 2023-01-07:
https://drive.google.com/open?id=1mdBgfmUMi2eAibjN7irGo4ariLNW_mtl&authuser=0&usp=drive_link

samspencer5991, 2023-01-07:
Have you tried refreshing your cache by pressing Ctrl + F5?

JiriHavlicek, 2023-01-07:
I tried. The entire browser froze and then completely crashed.

JiriHavlicek, 2023-01-07:
I guess I need to know what I'm doing wrong when updating the firmware. Did I understand the procedure correctly? - Download the Firmware updater, then connect the TRS with the Flexiports jack, then turn on the device. In the app it appeared as recognized, when I clicked on "download and update", the message "Waiting for device" appears. I wait twenty minutes and nothing.

samspencer5991, 2023-01-07:
We are discussing two completely separate issues. And GitHub isn't the ideal place for this. Let's sort out your firmware updating first. Perhaps create a support topic in our Discord server so I can more easily work out what your issue is.

JiriHavlicek, 2023-01-07:
Discord is unfamiliar to me, but I tried entering a topic.

closed 2023-01-12 by samspencer5991