Pirate-MIDI/Pirate-MIDI-BridgeOS #122: Bridge6 - sending configs to device from editor.

mandofink, 2023-01-07:
**Device(please complete the following information):**
   Bridge6  FW v1.2.0. HW v1.0.1

**Describe the bug**

After editing, unable to send to configs to device without factory reset.

**To Reproduce**
Steps to reproduce the behavior:
1. make a simple change using editor
2. send to device
3. web editor locks up,  also bridge6 locks up
4.  Factory reset allows me to send edits to device one time.

**Expected behavior**
i expected my edits to send to device

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Have had this problem for months,  finally went back and started from a factory reset yesterday, hoping to eliminate whatever bug crept into my config files (loading from a clean slate every time takes a long time).   worked pretty good for sending a few configs,  then it started locking up again.

thanks
jason

mandofink, 2023-01-07:
Right after sending this report,  and after 2 more config edits and Factory resets,   it is now loading from from the editor.   I have made about 5 more edits, and send-to-device successfully.

mandofink, 2023-01-08:
Worked fine the rest of yesterday,   today,  not able to send to device.

[image]

mandofink, 2023-01-10:
Is this the proper place to send this kinda stuff?   followed the link on the editor....

samspencer5991, 2023-01-12:
Hey mate, very sorry for the silence here, we've been flat out. This is quite interesting. There has been another issue reported similar to this and I should be able to get hands on with that device and try to work out exactly what is causing this as it isn't represented in all devices. So sit tight until I can do that

mandofink, 2023-01-12:
Thanks Sam, i am able to make progress despite the send issue,  and glad its not only me.  Happy to have the editor now - as i slowly figure this stuff out, there is a lot of back and forth! Is there an introduction to ‘switch groups’ video out there yet?  Something to consider with all that free time ;)JasonSent from my <advertisement removed>

mandofink, 2023-01-28:
Since the last update,  I have not had my bridge6 lock up during loading from the editor!   I have done it a bunch, and it seems to be resolved.   Yay!

I am still having some problems, which seemed to predate the update - my hold messages dont always seem to send - I will continue trouble shooting and open a new help ticket when I can be more specific!   Thanks!

closed 2023-02-20 by simonaglover95