Pirate-MIDI/Pirate-MIDI-BridgeOS #125: sequential mode issues

imranino, 2023-01-10:
here are a few issues I'm having with the sequential mode:
- pendulum mode goes through all the steps, then only goes between the first and last step. The name implies it ought to go from the first step through all the steps to the last, then backwards through all the steps to the first.
- only works with 'always' send. if i try 'on secondary action', the switch simply doesn't respond
- using a smart message to go a step backwards displays the correct step but doesn't send the step's message

simonaglover95, 2023-01-27:
Hi @imranino

I finally got around to testing all these individually. The pendulum mode is, indeed, broken and will be fixed in the next update.

Send on secondary action also not working for me and has been confirmed as a bug.

The last point - smart messages. When starting from scratch (all default settings for sequential mode) another switch with a "go to sequential step" Smart message works great and the message is sent as it should be. Perhaps you were not able to get it to send because the "send on secondary action" is broken? 

If you can (with v1.2.1) show the smart message isn't working, let me know the steps to recreate it and we'll get it fixed. 

Cheers
Simon

imranino, 2023-01-29:
Hi Simon,  Thanks so much for addressing these! I haven’t had the chance to test these out yet since updating to 1.2.1. I’m travelling at the moment but will be home next week so I’ll get right back into it. Keep it up!Imran Sent from my iPhone

closed 2023-02-20 by simonaglover95

frenno, 2023-02-24:
Hi Simon, I saw you closed this, is it fixed in 1.2.2? I'm on 1.2.1 and "Decrement Step (Sequential) smart messages are not sending the messages (Labels are being scrolled ut no messages sent)

This is with FS1 primary - Sequential / Always / Forward / Repeat last 3 and FS1 secondary - Hold Momentary. With the smart message in FS1 Hold stack.

simonaglover95, 2023-02-24:
Yeah it's fixed in 1.2.2. - we were going to release a couple of days ago
but decided to fix two other bugs first. Sorry for any confusion! :)