Pirate-MIDI/Pirate-MIDI-BridgeOS #127: Bank Drop Down List doesn't change selected bank

sliberty, 2023-02-05:
**Device(please complete the following information):**
 - Device: Bridge 6
 - Firmware version: 1.2.1
 - Hardware version: 1.1.3

**Describe the bug**
Dropping the bank list down on the main screen and selecting a different bank DOES NOT change which bank is shown on this screen. The up and down arrow buttons next to the bank drop down list DOES work as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Drop down the list, and select a different bank

**Expected behavior**
Displayed bank should change to the one selected in the drop down list

**Screenshots**
[image]
In this screen shot, the drop down list currently displays bank 8.

**Additional context**
Add any other context about the problem here.

sliberty, 2023-02-05:
I tested in the beta web app AND the beta "installed" app, and it behaved the same way in both.

samspencer5991, 2023-03-05:
@Konnsalad

Konnsalad, 2023-03-05:
A fix for this was pushed a few weeks ago, can you give it a test for me please @sliberty and confirm that it's working as expected for you now.

closed 2023-03-07 by Konnsalad

sliberty, 2023-03-12:
Working as expected. Thanks for applying this fix.