Pirate-MIDI/Pirate-MIDI-BridgeOS #128: Device link doesn't seem to transmit switch group messages between B6 and B4

Eldritch-Tentacles, 2023-02-06:
Having a Bridge6(Master) and Bridge4(Satellite) device linked; Switch groups work perfectly for turning off other switches onboard each individual device but don't turn off switches on the other device... with the exception that Switch 1 on Bridge6 WILL turn off the switches of Bridge4. No others either way.
Banking changes on both devices work with Device Link 
I have Switch Groups "enabled" on both devices. 
All switches in Switch Group on both devices for the same bank are set;
Broadcast: Transmit + Receive
Respond To: On + Off
Response: Off

markfarrier, 2023-04-01:
Unsure if this is the same issue but it's close enough I feel like I might be adding a duplicate if I raised a new bug.

I have 2 BRIDGE6s - BRIDGE6 A and BRIDGE6 B device linked with bank sharing and switch group sharing enabled.  One of these is technically Master and one is Satellite but switching these does not change/invert the behavior - so I'll just call them A and B instead as the behavior (the footswitch that works with device linked ) is linked to the device itself and not the settings on the device. (such as whether it's Master or Satellite)  Switch groups on both devices are configured to preset mode. (respond off to any received messages, and all switches in the bank have been added to the same group - group 1)

If you have a footswitch enabled on BRIDGE6 A, the only footswitch on BRIDGE6 B which will toggle off the Master footswitches is footswitch 1. (pressing any other footswitch on B will not cause the selected preset on A to toggle off)  If you have a footswitch enabled on B, the only footswitch on A which will toggle off the Satellite footswitch is footswitch 2.

Eldritch-Tentacles, 2023-04-18:
Hey, 
Sorry for such a delay in responding but, finally, yes, that’s pretty much the same issue I’ve had with this. 
Pressing any switch on my Bridge6 (master) will turn off any other switch on the Bridge6.... but only Switch 1 will turn off switches on the Bridge4 (satellite). 
The opposite is true.... pressing any switch on the Bridge4 will turn off any other switch on the Bridge4.... but only Switch 1 will turn off switches on the Bridge6.

simonaglover95, 2024-05-28:
Fixed in BridgeOS 2.0 Release

closed 2024-05-28 by simonaglover95