Pirate-MIDI/Pirate-MIDI-BridgeOS #131: Global Aux messages not working in some banks

imranino, 2023-02-12:
- Device: Bridge 4
 - Firmware version: 1.2.1
 - Hardware version: v1.0

I have a 3FS in Flexi1. I have them all the switches (press and hold) assigned to something globally. These are to turn on the tuner on my stomp, bank up and bank down on the primary, and go to menu bank (smart message), change FS mode on the stomp and something else on FS3 that eludes me right now.

For some reason, these don't work in some banks. Can't think why, I've checked to see there aren't any bank aux messages (BTW, to these trump the global ones? That could be very useful), and the global messages show.

**To Reproduce**
Steps to reproduce the behavior:
assign messages to au switch
go to certain banks
press aux switch
nothing happens

**Expected behavior**
the most important is to be able to scroll the banks when needed. This is particularly important as I often accidentally bank down by inadvertently pressing fs1 and 2 on the b4. So being able to quickly correct that by banking back up is the expected behaviour.

**Screenshots**
[image]

simonaglover95, 2023-02-15:
@imranino If you can, please upload the config file and tell me which bank numbers are not working. 

and FYI, the global and bank FS messages are independent and should both be sent, not one or the other - but it's a cool idea - I'll put it in the feature requests.

imranino, 2023-02-15:
I keep the same messages globally - I would surely forget any bank aux messages without any screen. But it could be useful and would make a lot of sense (to me) if the bank messages overrides of the global message.

here's my config. oh, not the .json file? How do I get the config file?

also in the screenshot, this is what the aux messages tend to look like in the editor.

[image]

samspencer5991, 2023-04-13:
@simonaglover95 where are you on this?

closed 2023-04-13 by samspencer5991