Pirate-MIDI/Pirate-MIDI-BridgeOS #134: Switch groups are not shown

mlacerte02, 2023-02-19:
**Device(please complete the following information):**
 - Device: Bridge6
 - Firmware version: v1.2.1
 - Hardware version: v1.1.2

**Describe the bug**
On the "Dev-Editor" when you go on "Bank settings" and in "Switch groups", nothing is shown. When you send to the device if the file contain "Switch group" they are sended or imported. But impossible to see or modify it in the editor.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Dev-Editor
2. Go to Bank settings
3. Click on Switch groups
4. See error

**Expected behavior**
I expected to see the switch groups, switch the midi command.

simonaglover95, 2023-02-20:
Hi @mlacerte02 , I can't seem to find the problem you're talking about. It's all working fine for me. I can create a switch group inside the bank settings on the editor and it sends to the device. 

Or, I can import from the device and the switch groups and their members are shown in the bank settings. I'm going to close this now. And by the way, the development editor is no longer recommended to use - we just pushed all the updates to the live version at edit.piratemidi.com :)

closed 2023-02-20 by simonaglover95

mlacerte02, 2023-02-20:
Perfect !,

After posting this bug, I find this problem intermittent and only a display problem in the beta editor as it actually sends to the bridge and come back to the editor. I don't know if you launched new "beta versions" yesterday, but this problem and others I found appeared and disappeared. Now everything is ok.

I will stop using the beta editor and start using the official one!

Thanks for your good job!

Salutations,

Martin Lacerte