Pirate-MIDI/Pirate-MIDI-BridgeOS #140: "Ramp" LFO shape repeats, should only run once

frenno, 2023-02-21:
**Device(please complete the following information):**
 - Device: Bridge 6
 - Firmware version: v1.2.1
 - Hardware version: 1.0.1

**Describe the bug**
The "Ramp" shape in the Switch LFO settings should rise once and then stop. However it keeps repeating (and therefore is the same as the "Sawtooth" shape)

**To Reproduce**
1. Use a blank template, add a CC message to the Toggle On stack of FS1 in bank 0 (eg Ch1, Controller 0, value 0)
2. Enable the Switch LFO and choose the "Ramp" shape. Other settings are not important.
3. Save to device and press FS1
4. Using a MIDI monitor, observe that the LFO runs repeatedly instead of once

**Expected behavior**
When set to "Ramp" the LFO should only run once.

**Additional context**
Happens whether Trigger is set to Toggle or Hold

simonaglover95, 2023-02-22:
Hey @frenno I can't replicate this bug. When I set the LFO on FS1 via the editor or onboard, I get a single ramp and it stops. If you could email a video that'd be good. But first, if you can, try to factory reset your device if you didn't after the last firmware update.

frenno, 2023-02-22:
I already did a factory reset after running into this issue (before replicating with a new blank template config for this bug report).

Have made a quick video, sending through now.

I forgot to mention before but I also tried setting it on the device (not web editor), still the same result.

frenno, 2023-02-22:
Ok as per your email, I've done some more testing - I had previously tried multiple other step sizes but not step size=1. And it turns out it only works for me when step size is 1 - I have updated the title to reflect this

samspencer5991, 2023-03-05:
Fixed in v1.2.2

closed 2023-03-05 by samspencer5991