Pirate-MIDI/Pirate-MIDI-BridgeOS #144: Sequential mode problems, over 9 steps programmed

MarkFSI, 2023-02-23:
**Device:**
 - Device: Bridge6
 - Firmware version: v1.2.1
 - Hardware version: v1.1.3

**Description of the bug**
in sequential mode only the first nine steps work. The others cannot be programmed from the device, and the programming of the first nine ones will also be compromised using the web editor, if the limit of 9 steps is exceeded.

**To Reproduce**
Steps to reproduce the behavior:
1. through the web editor, set a switch in sequential mode and add 9 steps with one message per step. Then send to device and check : all works properly
2. now try to add even just 1 step with one message and send to device. Than check the behaviour of the switch and its settings by the menu on the device: the settings of the steps and messages are corrupted

**Expected behavior**
A clear and concise description of what you expected to happen.

simonaglover95, 2023-02-24:
Hi Mark, can you tell me how many messages are assigned to each step? 
There is a limit of 16 messages total, no matter how many are assigned to each step. So if all 16 messages are used on those 9 steps, another step will not be able to be used. 

But it sounds like it might not be this problem, I just thought I'd mention it. If you can please give exact step to reproduce this problem, that would be helpful. Screenshots and photos are appreciated in more complex step-by-step processes like this one. You can also send your exported config file from the editor if you want so we can test it.

MarkFSI, 2023-02-24:
> But it sounds like it might not be this problem, I just thought I'd mention it.

Yes, you're right, I know what you have reminded but it's not this the problem.

MarkFSI, 2023-02-24:
> There is a limit of 16 messages total, no matter how many are assigned to each step. So if all 16 messages are used on those 9 steps, another step will not be able to be used.
> 
> But it sounds like it might not be this problem, I just thought I'd mention it.

Exactly, I know but the problem is not this.

In my test, I've assegned only one program change message for each step. Now I attach 2 files;
Please note that I've started labelling steps from number 0 and not 1:
[Sequential tests.zip](https://github.com/Pirate-MIDI/Pirate-MIDI-BridgeOS/files/10824608/Sequential.tests.zip)

P.S. is there a way to reply without change the state of the issue?

samspencer5991, 2023-03-05:
Fixed in v1.2.2

closed 2023-03-05 by samspencer5991