Pirate-MIDI/Pirate-MIDI-BridgeOS #162: LFO Saw weirdness

frenno, 2023-03-09:
**Device:**
 - Device: Bridge 6
 - Firmware version: 1.2.2 (with factory reset after upgrade from 1.2.1)
 - Hardware version: 1.0.1

**Describe the bug**
The Saw shape of the LFO should run continuously with the set step size and from the configured min to max. However under various circumstances this does not work.

I have provided a JSON file with various different issues shown on each footswitch in Bank 0. FS1 has the saw running at 1hz with step size 1, others are 0.5hz with step size 8 (bit easier to see the effects like this, but it is not happening because of the larger step size). Each footswitch has an LFO configured to use its Toggle On stack, which has a single CC in it.

When configured to use the full range (min: 0, max: 127) the saw only runs for half the period and then waits for half the period (FS1 and FS2 in the attached JSON). On FS2 due to the slower speed you can see that the LED is not lit fully when the sent value reaches max: it takes the full period of 2s (at 0.5hz) to fully light, whereas the messages increase up to 127 after 1s and then nothing is sent until it starts from the minimum again after waiting another 1s.

FS3 is configured with min:0, max:64 with step size 8. It runs up to 127 with step size 4 (???) but over the full period (no pause)

FS4 is configures with min:0, max:96, step size 8. It runs up to 127 with a step size of 6 (???) with roughly a 700ms pause (sending values over ~1300ms).

From these previous 2 examples it appears the step size is proportional to the fraction of the range 0-127 that the LFO is configured to: 4 is half  of 8, 64 is ~half of 127; 6 is ¾ of 8, 96 is ~¾ of 127. Not sure about the correlation of pause times though.

FS5 and FS6 are both configured with min:64, max:127, step size 8. The Toggle on message in FS5 has value 64, the one in FS6 has value 0. Neither of them work/send anything at all (although the LED lights as expected)

**Additional context**
Possibly related to fixes for #141  

[SawTest.zip](https://github.com/Pirate-MIDI/Pirate-MIDI-BridgeOS/files/10933954/SawTest.zip)

frenno, 2023-03-10:
After some more testing today:

- the Ramp shape behaves similarly to saw

- Triangle step size is scaled the same way (eg step size 6 when 8 is set on min:0-max:96)

The attached (zipped) JSON is identical to the SawTest one above but Bank 1 contains a copy with Ramp shape on each LFO and Bank 2 contains a copy with Triangle shape on each LFO. The three banks are otherwise identical.
[SawRampTriTest.zip](https://github.com/Pirate-MIDI/Pirate-MIDI-BridgeOS/files/10938252/SawRampTriTest.zip)

samspencer5991, 2023-03-22:
Fixed with v1.2.3
Note that step size is compressed with min/max thresholds accordingly

closed 2023-03-22 by samspencer5991