Pirate-MIDI/Pirate-MIDI-BridgeOS #163: Boot Delay

ChuckBarnard, 2023-03-10:
**Device(please complete the following information):**
Device: Bridge6
Firmware version: 1.2.2
Hardware version: 1.0.1

**Describe the bug**
The boot delay has stopped working with the last update.  System boots up without delay.

**To Reproduce**
Steps to reproduce the behavior:
I have the boot delay set to 20 seconds in the editor.  Haven't found the setting on the device itself.

**Expected behavior**
Need the option to send commands after everything else has started up.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

simonaglover95, 2023-03-10:
Hi @ChuckBarnard you can find the setting in ”Menu > global > interface > general UI” then scroll across to the second page. This info is also available in the user manual. 

Here, you can set/save the boot delay. My guess is a factory reset would fix it and then it will work from the editor, but you'd have to test and let me know. But setting onboard will probably be enough to kick it into working properly. Although I will test from my end today.

ChuckBarnard, 2023-03-10:
I have now verified the setting is correct on the device as well as I've completed a factory reset.  After importing my file there is still no boot delay.

samspencer5991, 2023-03-22:
Fixed with v1.2.3

closed 2023-03-22 by samspencer5991