Pirate-MIDI/Pirate-MIDI-BridgeOS #164: Screen Display

deandonaldson, 2023-03-11:
In the web editor it shows one less character compared to what you can actually have on the device.

Though it scrolls through, be better to extend the middle screen by one character to accurately reflect the device.

samspencer5991, 2023-03-27:
Hey @deandonaldson I can't replicate this. Are you saying that you can only enter in 15 characters on the web editor compared to the 16 you are allowed?

deandonaldson, 2023-03-28:
Hey @samspencer5991 

When you type into the editor it scrolls for the last character. You can physically enter it, but visually it doesn't show. It threw me originally as thought I was out of characters until I realised it wa just a design bug. 

I guess if you reduce font size a couple of points or enrage screen area on editor it would correct this?

closed 2023-04-13 by samspencer5991