Pirate-MIDI/Pirate-MIDI-BridgeOS #172: Sequential Note date does not save correctly

MIDISYNTHNOTE, 2023-03-20:
**Device(BRIDGE 6):**
 - Device:  [Bridge6]
 - Firmware version: [e.g. v1.2.2]
 - Hardware version: [e.g. v1.1.3]

**Describe the bug**
When you program Midi note data into one footswitch that is set to sequential the last note does not turn off even though I program it to turn off that specific note, it does not turn off the last note in the sequence. My Device would not save the info I programed in until I deleted the first step no matter how many times I tried to save it. It's strange because it works great with 4 steps but will not work with 7 or 8 steps.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to... Switch messages
2. Click on... Sequential
3. Program 7 to 8 steps with 2 MIDI notes for each step so the last note turns off when you press the footswitch (it would also be nice if switch groups worked with sequential steps, but they do not work when sequential is selected.

**Expected behavior**
It should save the data correctly

**Screenshots"

MIDISYNTHNOTE, 2023-03-21:
hello this is the same person that made the report for the bug.... I noticed that my bridge will not save anything now. I wonder if it has something to do with my device. everything saved and worked perfectly until today. The only thing I changed were the colors of the foot switches.

samspencer5991, 2023-03-21:
Hey mate,

I highly doubt it’s something to do with your device specifically. You
didn’t mention in your issue whether the config was being done onboard your
device or in the editor. Can you please update it with that info?

MIDISYNTHNOTE, 2023-03-21:
Okay After shutting down the unit for a while and coming back now it is saving most things correctly... except for the sequential midi note data. If I try to change or edit the 1st step on sequential MIDI note data it does not work even if the editor says that it's saved.

for the record.... I tried to save note data that I edited and the application says that it's saved and I unplug my machine and replug it back into a synthesizer and it reverts back and the data does not get saved.

MIDISYNTHNOTE, 2023-03-21:
Yes I'm doing all the programming in the editor.

I will update GitHub right now

MIDISYNTHNOTE, 2023-03-21:
I didn't think of it until you asked but I tried to program sequential notes into the actual unit and not use the web editor. and it worked! It would be nice if the web editor worked though. because I'm programming in noye data it takes an immense amount of time. 

side note: There's a few things I've noticed You can program directly into the machine but they are not on the web editor. for example TRS auxiliary foot switch bank change cannot be programmed in the web editor without changing settings in the machine itself.

There's a lot of back and forth not knowing If I should be programming it into the unit itself or if it can be done on the web editor. It would be great if the web editor and the unit itself had the exact same options.

Thanks again for your time.

samspencer5991, 2023-03-22:
That’s good to know as it helps dial in exactly where the error is
occurring.

The editor and onboard settings are both identical, and if something is
missing, it’s a bug, not by intention.

The reason I asked isn’t because they should behave differently, but to get
as much information on the issue as possible.

MIDISYNTHNOTE, 2023-03-22:
I didn't think of it until you asked but I tried to program sequential notes into the actual unit and not use the web editor. and it worked! It would be nice if the web editor worked though. because I'm programming in noye data it takes an immense amount of time. 

side note: There's a few things I've noticed You can program directly into the machine but they are not on the web editor. for example TRS auxiliary foot switch bank change cannot be programmed in the web editor without changing settings in the machine itself.

There's a lot of back and forth not knowing If I should be programming it into the unit itself or if it can be done on the web editor. It would be great if the web editor and the unit itself had the exact same options.

Thanks again for your time...

samspencer5991, 2023-04-14:
I've tracked down the cause of this bug and will have a fix out for it next week.

closed 2023-07-25 by simonaglover95