Pirate-MIDI/Pirate-MIDI-BridgeOS #173: Aux bank messages not working with Sequential (increment) (decrement) smart message.

MIDISYNTHNOTE, 2023-03-21:
I'm using the Web Editor only. The reason I use the web editor only is because it takes way less time programming, especially for note data sequentially like I'm doing.

 - Device: [e.g. Bridge6]
 - Firmware version: [e.g. v1.2.2]
 - Hardware version: [e.g. v1.1.3]

**Describe the bug**
Aux bank messages do not work with an external footswitch with Sequential (increment) (decrement) smart message. They do not trigger note data that is on the sequential  setting that's programed into a footswitch. When you program sequential decrement or increment auxiliary smart messages they work exactly like scrolling for note data right now. An external foot switch can't engage the note data sequentially.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to... bank settings aux messages 
2. Click on... tip or ring, press messages or toggle on/off messages, smart messages, sequential increment or decrement.

**Expected behavior**
aux sequential messages trigger note data just like on a footswitch of the bridge6. Right now it's acting exactly like scrolling.

I was thinking maybe there's a different way to make this work on the bridge 6... like send (note on) MIDI data with press and (toggle on) aux smart message sequential increment message at the same time. But nothing I've tried has worked so far. maybe I'm missing something and there's a different way to do this. Thank you very much for all your help and amazing customer service so far, I really appreciate it.

samspencer5991, 2023-04-13:
No issues with this on v1.3.0 for me. Please backup/update to the latest version of BridgeOS and if you're still having issues, shoot me your config.

closed 2023-04-14 by samspencer5991

MIDISYNTHNOTE, 2023-04-19:
Hello I was updating the firmware and my power went out. I tried to turn
the bridge six back on and it is not turning on at all. Is there something
I can do to fix this problem. I tried holding down footswitch number one
while turning on and nothing happened and I tried holding down footswitch
number six and the device will not power on