Pirate-MIDI/Pirate-MIDI-BridgeOS #176: Can't disable both bank triggers

GriffinSauce, 2023-03-26:
**Device(please complete the following information):**
 - Device: Bridge6
 - Firmware version: v1.0.1
 - Hardware version: v1.2.4

**Describe the bug**
It's not possible to disable both bank triggers:

**To Reproduce**
Steps to reproduce the behavior:
- Set both bank up and down triggers to "None"

**Expected behavior**
It disables both triggers.

**Screenshots**
[image]

**Additional context**
I have a dual aux switch to do bank switches, it should be possible to disable the triggers on the device completely.
This might just be an editor input-validation bug.

samspencer5991, 2023-03-26:
Absolutely just an input validation error. As you would expect, it won't let you make both triggers the same thing, but there obviously needs to be an exception for the none option. @konnsim

konnsim, 2023-03-26:
@GriffinSauce thanks for picking that up, just pushed a hotfix.

closed 2023-03-26 by konnsim

GriffinSauce, 2023-04-02:
Thanks @konnsim, works now!