Pirate-MIDI/Pirate-MIDI-BridgeOS #178: Web editor doesn't import from device with new firmware 1.3.0

MarkFSI, 2023-04-05:
**Device:**
 - Device: Bridge6
 - Firmware version: 1.3.0
 - Hardware version: 1.1.3

**Description of the bug**
With this new firmware version, web editor it's unable to import the configuration from device. When I click on "import from device", after selecting my bridge6, it is reported to me the following message: "Error: Connecting and loading data failed - Invalid value 0 for type false".
I've tried Google chrome and Microsoft edge, I've also changed PC, but it never works.

I hope you can solve this problem soon, very thanks in advance

MarkFSI, 2023-04-06:
Now the problem seems to be solved!
There is only a remaining odd behaviour:
with just opened editor, when you import configuration from device or from a file (saved with 1.3.0. firmware version) the editor firstly shows inverted colors for the text in the display of the device:
[image]
(check it selecting the name of switches or bank).
I've noticed that clicking on "bank settings" button, again on "bank settings" on the left and than on "discard & close" or "save & close" will check the problem temporarily, until you keep app open.

samspencer5991, 2023-04-08:
@konnsim

Glennmack, 2023-04-10:
@MarkFSI I'm having the same problem Mark. It seems Sam is aware of the issue and will focus on it this week. I'm not convinced it's a firmware issue, I rolled back to an older firmware and still have the same problem - suspect it's a Web Editor issue. Cheers, Glenn

konnsim, 2023-04-11:
Pushed a fix for the white boxes.

closed 2023-04-13 by samspencer5991