Pirate-MIDI/Pirate-MIDI-BridgeOS #179: Web Editor Update - Unable to Connect Bridge4

Glennmack, 2023-04-08:
**Device(please complete the following information):**
 - Device: Bridge4
 - Firmware version: v1.2.2
 - Hardware version: [e.g. v1.0]

**Describe the bug**
I went to =connect my iMac to the Bridge4 within Chrome. On connection, I was prompted to use a newer version of the web editor. I accepted, then I get the following error message. Computer restart didn't fix the issue. 

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
Add any other context about the problem here.

samspencer5991, 2023-04-08:
There have been 3 newer firmware versions since v1.2.2. Please ensure you are running the latest version of BridgeOS before creating an new issue.

Glennmack, 2023-04-08:
Hi Sam

My problem is that I can’t seem to get the Bridge4 to connect at all - no problems until today. I’ve installed the latest Updater but it doesn’t see the Bridge4 at all.

Cheers

Glenn

samspencer5991, 2023-04-08:
Ok no worries, that's a completely separate issue though.

We're working on a fix where the updater won't recognise devices with a Custom device name. So either factory reset your device, or reset your device name to Bridge6 or Bridge4 (try with and without the spaces), and you shouldn't have any issues. You can always update your firmware using the manual method by following the instructions on our website on the firmware page.

Glennmack, 2023-04-08:
Hi Sam

I needed to reset the unit, renaming didn’t work as it kept reverting to the custom name, even though it said it saved the change.

Another problem - I now get this screen. I have the latest version ion Chrome installed, the latest firmware and the latest Updater.

Any ideas?

Glenn

samspencer5991, 2023-04-08:
Your picture hasn't come through, but I'm going to assume that you just need to perform the factory reset as required after the firmware update :)

Glennmack, 2023-04-08:
Factory Reset was completed. Unfortunately I’m back to my original problem.

samspencer5991, 2023-04-08:
Ok, could you please open to chrome console (F12) and copy the contents of the console when this occurs?

Glennmack, 2023-04-08:
I’m not familiar with Chrome Console - can’t find anything on the Mac. Any tips?

samspencer5991, 2023-04-08:
All good, press F12 and a little window will open up on the right. Click on the console tab and copy the contents

Glennmack, 2023-04-08:
Here tis, I needed to enable standard function keys:

:

samspencer5991, 2023-04-08:
I can't see any console logs attached mate, could you confirm please?

Glennmack, 2023-04-08:
Screenshot:

Glennmack, 2023-04-08:
[image]

samspencer5991, 2023-04-08:
@konnsim

Glennmack, 2023-04-08:
I'm done for the night Sam. The latest is I used an older Web Editor link and got as far as trying to update the now rest Bridge4. It gets as far as 15% and then freezes like this. 

[image]

samspencer5991, 2023-04-08:
No worries mate, we'll get this sorted out asap for you. Could you please export your config file from the editor and email it to us and I'll get it looked at first thing next week

Glennmack, 2023-04-08:
Thanks Sam

I've had a little success, this involved reverting to Firmware v1.2.2. 

I then managed to get the Web Editor to open my backup file, I've tried to attach this below but it seems the file type isn't accepted here - I'll try emailing it. When I loaded this file, the names of all of the switches and banks were whited out as shown above, but by going into Global settings and out again, the text all appeared as normal (i.e. white text on dark background). However, I couldn't export this file back to the Bridge4. It kept sending up the error message.

To overcome this, I opened Web Editor in two windows, I copied each bank (bank by bank) to the clipboard, then pasted it into a new file. I was then able to send this new .json file to the Bridge4. 

So, I am now set up with v1.2.2, have managed to rescue my many banks (a lot of time went into these so I'm relieved), and have saved as a new exported .json file. Note though, when I load this file ('Import From File') the text is again whited out until I go into Global Settings and out again. 

Cheers

Glenn

Glennmack, 2023-04-08:
Sam

Here is the exported file that I’ve been having trouble with.

Cheers

Glenn

Glennmack, 2023-04-08:
One last update - what I thought was success isn't actually success. The copied file has the same issues.

konnsim, 2023-04-11:
Pushed a fix for the white boxes.

Glennmack, 2023-04-11:
Thanks Kurtis. Will this also fix the issue related to connection to Bridge4? I’ve been unable to upload from and send to Bridge4 from the web editor since the latest update. Thanks.

samspencer5991, 2023-04-13:
@Glennmack the white boxes was entirely unrelated mate, that was just a UI bug in the editor.

Coming back to your original issue. You've backed up your config, so that's good. I haven't received an email from you, nor is your config attached here. Can you please make sure you have sent it to sam@piratemidi.com

Glennmack, 2023-04-13:
Hi SamThanks - I’m away this week so will send through by email tomorrow night. CheersGlenn

Glennmack, 2023-04-15:
I've rebuilt my entire Bridge4 settings to overcome this issue.

closed 2023-04-15 by Glennmack