Pirate-MIDI/Pirate-MIDI-BridgeOS #181: Scrolling Linked Mode - Issue With Switches 2 & 4

Glennmack, 2023-04-15:
**Device(please complete the following information):**
 - Device: Bridge 4
 - Firmware version: v1.0.1
 - Hardware version: v1.3.0

**Describe the bug**
I'm using Scrolling Linked Mode on switches 1 & 3 and switches 2 & 4.
Switch 1 & 3 work fine, the label changes on both switches when scrolling either direction 
Switch 2 & 4 are linked in exactly the same way, the babel only changes on switch 2, switch 4 label stays at 0.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Expecting switch labels to both changed as per switch 1 & 3 in Scrolling Linked Mode.

**Screenshots**
Attached

**Additional context**
Add any other context about the problem here.
[image]

samspencer5991, 2023-04-17:
Can you please upload your config here for me to look at.

Glennmack, 2023-04-17:
Hi Sam

This is the old config - I was unable to get this working again since the firmware upgrade. 

This is my new config which is working fine but needed to be set up from scratch.

Cheers

Glenn

samspencer5991, 2023-04-17:
Ok great, please make sure that if you're creating an issue and then it is solved in a firmware update that you close the issue if I haven't already.

Glennmack, 2023-04-17:
Hi Sam

I don’t believe that it was solved - it required me starting from scratch. Also, I did close the issue over the weekend.

Cheers

Glenn

samspencer5991, 2023-04-17:
Ah ok, a bit of a miscommunication regarding the purpose of solving issues. 'Solving' an issue doesn't refer to a customer service perspective. It means that the problem/bug you were experiencing is no longer a problem/bug due to a firmware update, hardware revision etc... (speaking generally there).
I would be more than happy to look at your original config and work out what wasn't working (I'm assuming that template was brought over from a previous firmware version?) but given you only created this issue 2 days ago (over the weekend), there hasn't really been a realistic timeframe to do that with a config which I had to request.

Maybe GitHub is wigging out here but I can't see any action showing the issue being closed prior to me closing it a few minutes ago.

But glad you're up and running with that particular setup. Could you tell me a bit more about why you had to start from scratch and how that fixed it? Did just reassigning the switches accordingly for whatever bank you're using not sort things out?

samspencer5991, 2023-04-17:
Closed is probably a better term here than solved, but the outcome is the same :)

Glennmack, 2023-04-17:
Hey SamThe issue started when I connected to the web editor and I couldn’t import from device. The import kept freezing at 18% and then came up with an error message. This is when I first logged a bug report. Then either you or Simon pointed out that there was a new firmware and that may have been the reason I had the issue. So, I updated the firmware, reset the Bridge4, then had the same issue. At the time I recall thinking that it could have been simply a web editor issue. From there I loaded the older firmware again, same issue. I couldn’t even load my old saved config back into the Bridge4. During that whole process, there was a fix pushed to the web editor to overcome a white text on white box issue. I don’t think that was relevant to my specific problem though. There was a lot of to and fro’ing around configs etc, then I chose to load the new firmware, reset the Bridge4 again and built from scratch as I wanted to get back to using it. Once I’d done that, there were no further issues with importing from the Bridge4 or sending to it. Hope that helps in some way. CheersGlenn

samspencer5991, 2023-04-17:
Ah ok gotcha. The White text box issue was entirely separate and not related to this at all.
Well, sorry for the hassle, but glad it's now sorted. All the best!

Glennmack, 2023-04-17:
Hey Sam

I had a thought - I’ve just checked and my thought was correct. We are talking about two completely different issues that I’ve logged. My previous email was all in relation to an earlier issue - that’s the one I closed on the weekend and was overcome by rebuilding the Bridge4.

The one regarding scrolling not being linked between switches 2 and 4 is still unresolved, even with the firmware update.

I’ve added a pic and a short video showing this below. 

Sorry for the confusion - I was thinking some of it didn’t make sense.

Cheers

samspencer5991, 2023-04-17:
Ok no worries, I'll open this back up again. Could I please get your config for this issue?

Glennmack, 2023-04-17:
Hi Sam

Is that the .json file? I’ve attached below.

Cheers

Glenn

Glennmack, 2023-04-17:
Hi Sam

Not sure if the files came through via email - I've added them here. I changed the .json to a .txt file 0 seems json can't be pasted in github.

Cheers and thanks for your patience and help!

Glenn

[image]

https://user-images.githubusercontent.com/128812448/232466542-6f9d980b-9968-4310-b4bd-8549df4c3587.mov

[Bridge 4_2023-04-16_10-36-01 copy.txt](https://github.com/Pirate-MIDI/Pirate-MIDI-BridgeOS/files/11249406/Bridge.4_2023-04-16_10-36-01.copy.txt)

simonaglover95, 2023-04-19:
I've confirmed this bug on both BRIDGE6 and BRIDGE4 in the latest firmware (v1.3.0) today. It happens with a BRIDGE6 when FS 3&6 are linked just like the BRIDGE4 when FS2&4 are linked.

simonaglover95, 2023-04-26:
This is fixed in the v1.3.1 patch which will be released shortly.

Glennmack, 2023-04-26:
Awesome, thanks Simon.

closed 2023-07-25 by simonaglover95