Pirate-MIDI/Pirate-MIDI-BridgeOS #182: Bridge 4 keeps freezing when using the web editor

victorcamnerin, 2023-04-17:
When using the web editor and trying to `Send to device` my Bridge 4 freezes. It's very obvious that it happens when the computor is trying to connect to the device as the tap tempo led stops blinking on the device. Once it's frozen i can't do anyting on the Bridge. Only solution is to cut the power.

I attach a screenshot of the error message I get.

[image]

Here's the console messages in chrome. 

[image]

victorcamnerin, 2023-04-17:
Also the firmware updater don't seem to recognise my device. I never get anyting but the start screen there with the different devices displayed.

simonaglover95, 2023-04-19:
Hi Victor, what firmware version are you on? I assume it's not up to date seeing as you are looking at the firmware updater? 

Try this: Reboot your computer, then plug the BRIDGE in, then open the firmware updater without doing anything else first (don't try use the editor, don't touch any switches on the BRIDGE). Wait a bit for the updater app to check for updates, and click "Yes" if there's an updater. Wait for the app to finish the update and ask you to restart it. When the app has been restarted after the update, wait about 30 seconds for the app to look for the BRIDGE device. If it isn't found, unplug the USB, plug it in again, and wait another 30 seconds. Please make sure the 9v DC power is NOT plugged in while trying to use the USB. Sometimes the 9v can upset the process. 

Let me know how you go. Cheers.

victorcamnerin, 2023-04-29:
Hi @simonaglover95!

After a restart and an update of my mac I could manually update the Firmware. Now the updater recognizes the my Bridge 4. 

It seems the unit is working better with the editor as well.  However, sometimes I need to start the mac MIDI and sound settings menu to get the editor to reconize the Bridge. Not everytime, but when it doesnt show up that is a workaround at least.

Thanks for your help!

systemcode, 2023-06-25:
I have the same issue of the web editor and trying to Send to device my Bridge 6 freezes. Exactly as described above. 
Latest MacOs and Chrome up to date.
Bridge 6
1.3.1 Current Firmware Version
1.1.3 Hardware Revision

(My updater however works perfectly well and never fails to connect)

closed 2023-07-25 by simonaglover95