Pirate-MIDI/Pirate-MIDI-BridgeOS #187: Switch in momentary mode ,LED turned on incorrectly when using Smart message!

Dearxj3gk6, 2023-05-11:
**Device(please complete the following information):**
 - Device: Bridge4
 - Firmware version:   v1.3.1
 - Hardware version:  v1.1.3

**Describe the bug**
Switch in momentary mode , when press the switch send out smart message, Switch's LED will not turn on .
**To Reproduce**
Steps to reproduce the behavior:
1. Go to momentary mode 
2. Click on press message 
3. send any smart message  
4. See error

**Expected behavior**
The Switch in momentary mode , when press the switch send out smart message, Switch's LED will  turn on .

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

simonaglover95, 2023-05-16:
I've confirmed this bug and we're looking into it. Thanks for finding it! 👍

simonaglover95, 2024-05-28:
Fixed in BridgeOS 2.0 Release

closed 2024-05-28 by simonaglover95