Pirate-MIDI/Pirate-MIDI-BridgeOS #190: Scrolling Counter resets on BankDown / GoToBank

TheGuacamoleXplosion, 2023-06-08:
BRIDGE4
Firmware 1.3.1
Hardware 1.0.1

When creating a scrolling FS pair on any Bank, the scrolling counter will **only** maintain its current value when navigating to that Bank via BankUp. While navigating to that Bank by either BankDown or GoToBank will reset the scrolling counter. 

Desired behaviour: The scrolling counter will always remember its value when changing Banks by any means. Unless I deliberately reset it.

I found this when creating one Bank to scroll through program changes on my Line 6 M5, and a second Bank to scroll through the looper level CC on my Strymon Timeline.

Steps:

1. Start with the empty template in the editor.

2. Set Globals to 
Preserve Switch State = Yes,
Transmit Switch State = No,
Bank Up Trigger = FS 4 (also tested other triggers incl. Aux Switch)
Bank Down Trigger = FS 3 (also tested other triggers incl. Aux Switch)
and save.

3. On Bank 0, 
set FS1 to Scrolling, Always, Reverse, Interval=1, Min=0, Max=23
and FS2 to Scrolling Linked, Always, Forward, Linked FS=1.
and save.

4. On Bank 1, 
set FS1 to Scrolling, Always, Reverse, Interval=4, Min=0, Max=127
and FS2 to Scrolling Linked, Always, Forward, Linked FS=1.
and save.

5. On Bank 2,
set FS1 to Momentary, Press=Go-To Bank 0
and FS2 to Momentary, Press=Go-To Bank 1
and save.

6. Send to Device. 

7. Go to Bank 1. Scroll to, say, value 111.
Click FS3 to go down to Bank 0. Scroll to, say, value 17.
Click FS3 to go down to Bank 99.
Click FS4 to go back to Bank 0: Voilà value=17.
Click FS4 to go up to Bank 1: Voilà value=111.
Click FS4 a hundred times to go through all values. 17 on Bank 0 and 111 on Bank 1 are still intact.
Click FS4 to go up to Bank 2. 
Click FS3 to go back to Bank 0. Voilà value reset to 127.
Click FS3 to go down to Bank 0. Voilà value reset to 23.
Scroll to, say, value 5. 
Click FS4 to go up to Bank 1. Scroll to, say, value 8.
Click FS4 to go up to Bank 2.
Click FS1 to go straight to Bank 0. Voilà value reset to 23.
Click FS4 to go up to Bank 1. Voilà value 8 is still intact.
Click FS4 to go up to Bank 2.
Click FS2 to go straight to Bank 1. Voilà value reset to 127.

TheGuacamoleXplosion, 2023-06-08:
Addendum: 
Right after this, I wanted to test something else and started from a blank template in the editor again. Made a very simple thing:

On Bank 0,
set FS1 to Scrolling, Always, Reverse, Interval=4, Min=0, Max=127
and FS2 to Scrolling Linked, Always, Forward, Linked FS=1.
and save.

Sent to device. 

Result: I get a scrolling pair with values 0 to 23 in intervals of 4. In other words, the MAX value from the previous configuration persisted on the BRIDGE4, while the interval was successfully overwritten. 

Altered interval to 3, re-sent to device. Old MAX value of 23 is still in place.

simonaglover95, 2024-05-28:
Fixed in BridgeOS 2.0 Release

closed 2024-05-28 by simonaglover95